balena-os/jetson-flash #13: Non-flasher images are not supported

roman-mazur, 2019-09-18:
So it cannot be used for Jetson Nano

imrehg, 2019-09-18:
That link above is to our forums, here https://forums.balena.io/t/flashing-jetson-tx2-via-usb/2472/48, just for reference, and at least 2 people asked about it.

maxritter, 2019-09-22:
@roman-mazur There is an updated version from @telphan that in theory should support the Nano production: https://github.com/balena-os/jetson-flash/tree/fix
But I am not getting it to work properly. The flash.xml file has to be replaced by mine: https://forums.balena.io/t/flashing-jetson-tx2-via-usb/2472/63?u=maxritter, but the device is not booting. Let me know if you have an idea what could be the reason.

telphan, 2019-10-07:
Fixed by #14

closed 2019-10-07 by telphan