balena-os/jetson-flash #17: Expected to work with a Jetpack 4.2.2 image?

mhodgson, 2019-10-03:
Hi there, I tried to use this tool with a fresh TX2 image (Jetson 4.2/Warrior) generated off master of https://github.com/balena-os/balena-jetson.

The flashing over USB appeared to finish successfully, but the board was bricked (nothing on screen, no boot). 

Is this tool expected to work for this use case, or does it need to be updated to work with the latest version of Jetpack? Sorry, I'm just not familiar enough with the guts of this thing to debug it further myself.

Thanks!

telphan, 2019-10-07:
@mhodgson Yes, I believe the latest OS image and the version of L4T that this tool flashes are incompatible now. There is an open PR here #18 which should work with an image cut from master.

mhodgson, 2019-10-07:
Ah, I figured as much. Thanks for the response. Looking forward to the L4T 32.2 support!

closed 2019-10-07 by mhodgson