balena-os/jetson-flash #20: Flashing on the latest master fails

joshughes, 2020-03-02:
```
root@6865119:/data/jetson-flash# bin/cmd.js -p -o /data/output -f /data/resin-build-360.img -m jetson-tx2
Consistency check done for /data/output/jetson-flash-artifacts/Linux_for_Tegra
Using cached Linux_for_Tegra
Checking resin cache
Cache image check done
Retrieve resin-image from /data/resin-build-360.img
Writing to /data/output/jetson-flash-artifacts/resin/img [========================] 100% eta 0s  
Copying resin configuration from /data/resin-build-360.img to /data/output/jetson-flash-artifacts/resin/img
Extracting partition kernel from /data/output/jetson-flash-artifacts/resin/img to /data/output/jetson-flash-artifacts/resin
Extracting partition kernel_b from /data/output/jetson-flash-artifacts/resin/img to /data/output/jetson-flash-artifacts/resin
(node:1152) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'name' of undefined
    at Object.exports.extractPartition (/data/jetson-flash/lib/utils.js:137:38)
    at async ResinJetsonFlash.generateArtifacts (/data/jetson-flash/lib/resin-jetson-flash.js:228:14)
    at async ResinJetsonFlash.run (/data/jetson-flash/lib/resin-jetson-flash.js:248:3)
    at async run (/data/jetson-flash/bin/cmd.js:56:2)
(node:1152) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:1152) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

When I checkout v0.3.0 of the code this issue does not occur.

acostach, 2020-03-03:
Hi @joshughes , which repository did you use for your build? If you used 'balena-jetson-skx2' repository, then than one uses L4T 28.3, which is compatible up to the tag v0.3.0. The new 'balena-jetson' repository is compatible with L4T 32.2, and can be flashed with the master revision here. 

From now on, balena-jetson repository will be used for the jetson boards. I've also updated the readme in this PR: https://github.com/balena-os/jetson-flash/pull/21

acostach, 2020-03-25:
@joshughes do you have other questions related to the above? Can we close the ticket?

acostach, 2020-03-26:
@joshughes I assume the details above and the README updates fixed the problem you encountered, so I'll close this ticket for now. Feel free to re-open or ping me here if you still get those errors.

closed 2020-04-08 by joshughes