balena-os/jetson-flash #22: flashing balena-jetson fails with l4t-32.3.1

wremie, 2020-03-04:
flashing the image build from balena-jetson on latest master fails. Probably their have been some changes in partition layout between the two l4t versions. I've tried to switch to latest l4t sources but results in the same error.

```
./bin/cmd.js -m jetson-nano-emmc -f ../balena-jetson/build/tmp/deploy/images/jn30b-nano/resin-data.img -p -o flash
Consistency check done for /home/willem/balena/jetson-flash/flash/jetson-flash-artifacts/Linux_for_Tegra
Using cached Linux_for_Tegra
Checking resin cache
Cache image check done
(node:91834) UnhandledPromiseRejectionWarning: SyntaxError: Invalid MBR boot signature. Expected 0xAA55, but saw 0x0
    at Function.MBR.detectFormat (/home/willem/balena/jetson-flash/node_modules/mbr/lib/mbr.js:73:11)
    at Function.MBR.parse (/home/willem/balena/jetson-flash/node_modules/mbr/lib/mbr.js:54:20)
    at new MBR (/home/willem/balena/jetson-flash/node_modules/mbr/lib/mbr.js:12:16)
    at getPartitionsFromMBRBuf (/home/willem/balena/jetson-flash/node_modules/partitioninfo/build/partitioninfo.js:49:11)
    at /home/willem/balena/jetson-flash/node_modules/partitioninfo/build/partitioninfo.js:100:18
(node:91834) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:91834) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

acostach, 2020-03-04:
@wremie looks to me like you are trying to flash the 'resin-data' partition image
`../balena-jetson/build/tmp/deploy/images/jn30b-nano/resin-data.img`
and not the actual BalenaOS image.

The deployed image should look similar to:
`resin-image-jn30b-nano-20191216162223.rootfs.resinos-img`

wremie, 2020-03-04:
Probably, I can claim the most stupid mistake of the week allready......

acostach, 2020-03-05:
@wremie  can you please try do a clean build with https://github.com/balena-os/balena-jetson/pull/51 ? I've successfully flashed a jn30b nano emmc image on the devkit with it. If you are still getting build errors, try doing a clean build with a new sstate directory.

wremie, 2020-03-05:
Hi @acostach, I can confirm that the build succeeded on my side as well. The device (JN30B Rev. 4) flashed successfully as well. Besides, the DTB should be loaded correctly as the SD-card bug is solved compared to the old release. [balena-os/balena-jetson](https://github.com/balena-os/balena-jetson/pull/51) may be merged.

closed 2020-03-05 by wremie

acostach, 2020-03-05:
thanks for the confirmation @wremie . Please also update the BSP link on your side, from 32.2 to 32.3.1 here: https://github.com/balena-os/jetson-flash/blob/bf222ad8cc727a711c4832d9750dd68e38add4e2/lib/resin-jetson-flash.js#L63 I'll update the link also here after the image is released to production. You can find the link at https://developer.nvidia.com/embedded/linux-tegra as L4T Driver Package (BSP) under the Jetson Nano and TX1 category.

wremie, 2020-03-05:
@acostach I've succesfully reflashed the board with updated link and works like a charm.