balena-os/jetson-flash #23: Populate data dir with files

joshughes, 2020-04-08:
When we flash a device it would be great if there was a way to populate the `resin-data` partition with some files that we would want to ensure are there for the life of the robot. 

It seemed like this may be possible with some modifications to this tool. I wanted to ask if this is possible already or would be something that could be considered to be added. 

Thanks !

acostach, 2020-04-08:
@joshughes maybe https://www.balena.io/docs/learn/develop/runtime/#persistent-storage helps?

joshughes, 2020-04-08:
@acostach yes I may not be using the right terminology, but what I am talking about is that when we manufacture a device, we would like to flash that device and put some files on the persistent-storage. It seems like as we flash our application it would be a good time to use the flash process to put things like keys into this partition so on first boot the device can contact the services it needs.

acostach, 2020-04-09:
@joshughes are these files unique per device or per application? If they are to the application, then preloading the app in the image with 'balena app preload' is one way to do this.

acostach, 2020-10-27:
Closing due to inactivity, can be re-opened if necessary.

closed 2020-10-27 by acostach