balena-os/jetson-flash #27: Failed to flash Xavier NV 32_4_2

kmilo7204, 2020-08-03:
Balena Team, 

I will try to explain and describe our setup and the outcome that we got in order to make it clear.

We are currently trying to get BalenaOS in the Jetson Xavier NX Developer Kit. We have an image provided by your team (balena-staging-APP-jetson-xavier-nx-devkit-emmc-2.51.1+rev5-v11.4.10.img). According to your description this image should be able to flash the device on the EMMC with BalenaOS; however, we are facing some issues as we are not able to complete the process.

We followed the steps given in the README.md:

1. Xavier NX is connected via USB to a PC (OS: Ubuntu 18.04)
2. Xavier NX is in recovery mode.
3. sudo privileges are provided to run the Nvidia flasher
4. Version of BalenaOS is 2.51.1 (Which uses our current L4T - R 34.4.2)

After all the elements were ready we executed ./bin/cmd.js -f balena.img -m <device_type>. Everything was working good, but after some seconds it got stuck (See the shared logs below), we waited for almost one hour and we got a message like this one (CPU Bootloader is not running on device)

It is important to say that we tried different setups (Changing L4T as stated below), and even your two different branches in this repo (34_4_2 and 34_4_2-build) but the outcome was not different.

 * L4T JP 4.4 - R 34.4.3
 * L4T JP 4.4 - R 34.4.2

As additional information, I found some related Nvidia Forums but they were not helpful at all: 

 * https://forums.developer.nvidia.com/t/failed-to-flash-tx2/57024/4
 * https://forums.developer.nvidia.com/t/tx2-flashing-cpu-bootloader-is-not-running-on-device-error-return-value-4/58057/5
 * https://forums.developer.nvidia.com/t/cpu-bootloader-is-not-running-on-device/108313

> ` Generate header for tegra194-p3668-all-p3509-0000_sigheader.dtb.encrypt
Generate 16-byte-size-aligned base file for tegra194-p3668-all-p3509-0000_sigheader.dtb.encrypt
Remove temporary file
done.
Making system.img... 
	populating rootfs from /tmp/Linux_for_Tegra/rootfs ... done.
	Sync'ing system.img ... done.
	Converting RAW image to Sparse image... done.
system.img built successfully. 
Existing tbcfile(/tmp/Linux_for_Tegra/bootloader/cboot_t194.bin) reused.
copying tbcdtbfile(/tmp/Linux_for_Tegra/kernel/dtb/tegra194-p3668-all-p3509-0000.dtb)... done.
copying cfgfile(/tmp/17205/resin/flash.xml) to flash.xml... done.
Existing flasher(/tmp/Linux_for_Tegra/bootloader/nvtboot_recovery_cpu_t194.bin) reused.
Existing flashapp(/tmp/Linux_for_Tegra/bootloader/tegraflash.py) reused.
./tegraflash.py --bl nvtboot_recovery_cpu_t194.bin --sdram_config tegra194-mb1-bct-memcfg-p3668-0001-a00.cfg,tegra194-memcfg-sw-override.cfg  --odmdata 0xB8190000 --applet mb1_t194_prod.bin --cmd "flash; reboot" --soft_fuses tegra194-mb1-soft-fuses-l4t.cfg  --cfg flash.xml --chip 0x19 --device_config tegra19x-mb1-bct-device-qspi-p3668.cfg --misc_cold_boot_config tegra194-mb1-bct-misc-l4t.cfg --misc_config tegra194-mb1-bct-misc-flash.cfg --pinmux_config tegra19x-mb1-pinmux-p3668-a01.cfg --gpioint_config tegra194-mb1-bct-gpioint-p3668-0001-a00.cfg --pmic_config tegra194-mb1-bct-pmic-p3668-0001-a00.cfg --pmc_config tegra19x-mb1-padvoltage-p3668-a01.cfg --prod_config tegra19x-mb1-prod-p3668-0001-a00.cfg --scr_config tegra194-mb1-bct-scr-cbb-mini-p3668.cfg --scr_cold_boot_config tegra194-mb1-bct-scr-cbb-mini-p3668.cfg --br_cmd_config tegra194-mb1-bct-reset-p3668-0001-a00.cfg --dev_params tegra194-br-bct-qspi.cfg  --bin "mb2_bootloader nvtboot_recovery_t194.bin; mts_preboot preboot_c10_prod_cr.bin; mts_mce mce_c10_prod_cr.bin; mts_proper mts_c10_prod_cr.bin; bpmp_fw bpmp_t194.bin; bpmp_fw_dtb tegra194-a02-bpmp-p3668-a00.dtb; spe_fw spe_t194.bin; tlk tos-trusty_t194.img; eks eks.img; bootloader_dtb tegra194-p3668-all-p3509-0000.dtb"   
saving flash command in /tmp/Linux_for_Tegra/bootloader/flashcmd.txt
saving Windows flash command to /tmp/Linux_for_Tegra/bootloader/flash_win.bat
*** Flashing target device started. ***
Welcome to Tegra Flash
version 1.0.0
Type ? or help for help and q or quit to exit
Use ! to execute system commands `
 
> `[   0.0016 ] tegrasign_v2 --key None --getmode mode.txt
[   0.0025 ] Assuming zero filled SBK key
[   0.0027 ] 
[   0.0028 ] Generating RCM messages
[   0.0044 ] tegrahost_v2 --chip 0x19 0 --magicid MB1B --appendsigheader mb1_t194_prod.bin zerosbk
[   0.0051 ] Header already present for mb1_t194_prod.bin
[   0.0080 ] 
[   0.0088 ] tegrasign_v2 --key None --getmode mode.txt
[   0.0095 ] Assuming zero filled SBK key
[   0.0096 ] 
[   0.0104 ] tegrasign_v2 --key None --file mb1_t194_prod_sigheader.bin --offset 2960 --length 1136 --pubkeyhash pub_key.key
[   0.0111 ] Assuming zero filled SBK key
[   0.0117 ] 
[   0.0127 ] tegrahost_v2 --chip 0x19 0 --updatesigheader mb1_t194_prod_sigheader.bin mb1_t194_prod_sigheader.hash zerosbk
[   0.0161 ] 
[   0.0177 ] tegrabct_v2 --chip 0x19 0 --sfuse tegra194-mb1-soft-fuses-l4t.cfg sfuse.bin
[   0.0234 ] 
[   0.0262 ] tegrarcm_v2 --listrcm rcm_list.xml --chip 0x19 0 --sfuses sfuse.bin --download rcm mb1_t194_prod_sigheader.bin 0 0
[   0.0286 ] RCM 0 is saved as rcm_0.rcm
[   0.0385 ] RCM 1 is saved as rcm_1.rcm
[   0.0389 ] RCM 2 is saved as rcm_2.rcm
[... 389 lines cut ...]
[   7.1039 ] Added binary blob_mts_c10_prod_cr_sigheader.bin.encrypt of size 3418256
[   7.1048 ] Added binary blob_bpmp_t194_sigheader.bin.encrypt of size 856112
[   7.1057 ] Added binary blob_tegra194-a02-bpmp-p3668-a00_sigheader.dtb.encrypt of size 172720
[   7.1061 ] Added binary blob_spe_t194_sigheader.bin.encrypt of size 94960
[   7.1064 ] Added binary blob_tos-trusty_t194_sigheader.img.encrypt of size 373696
[   7.1067 ] Added binary blob_eks_sigheader.img.encrypt of size 5136
[   7.1070 ] Added binary blob_tegra194-p3668-all-p3509-0000_sigheader.dtb.encrypt of size 201072
[   7.1093 ] 
[   7.1094 ] Sending bootloader and pre-requisite binaries
[   7.1103 ] tegrarcm_v2 --download blob blob.bin
[   7.1111 ] Applet version 01.00.0000
[   7.1282 ] Sending blob
[   7.1286 ] [................................................] 100%
[   7.9193 ] 
[   7.9219 ] tegrarcm_v2 --boot recovery
[   7.9243 ] Applet version 01.00.0000
[   7.9475 ] 
[   8.9496 ] tegrarcm_v2 --isapplet
[ 1022.7454 ] 
[ 1022.7482 ] tegrarcm_v2 --ismb2
[ 2038.5559 ] tegradevflash_v2 --iscpubl

<summary>

Is there something that we are missing?

acostach, 2020-08-04:
HI @kmilo7204 

I checked the 32_4_2 branch now and flashed the NX successfully twice. My setup consists of the NX module plugged in a B01 Nano carrier board.

The command I used was:

> ./bin/cmd.js -f balena-staging-MYAPP-jetson-xavier-nx-devkit-emmc-2.51.1+rev5-dev-v11.4.10.img -m jetson-xavier-nx-devkit-emmc

I assume you used the 'jetson-xavier-nx-devkit-emmc' machine in the cmdline too, and lsusb shows the Xavier in recovery mode, correct?

If so, can you please try flashing only the tegra bootloaders with these steps, see if it works?

> wget https://developer.nvidia.com/embedded/L4T/r32_Release_v4.2/t186ref_release_aarch64/Tegra186_Linux_R32.4.2_aarch64.tbz2 && tar xf Tegra186_Linux_R32.4.2_aarch64.tbz2 && cd Linux_for_Tegra && sudo ./flash.sh jetson-xavier-nx-devkit-emmc mmcblk0p1

JohnBetaCode, 2020-08-04:
Wait, why are you using a  B01 Nano carrier board with a NX module?  

We have been reading a lot and this version that we bought https://developer.nvidia.com/embedded/jetson-xavier-nx-devkit it seems that does not have eMMC. Please tell us, what board we should use or if there's another NX version with eMMC.

JohnBetaCode, 2020-08-04:
We found this in an Nvidia page:
** The Jetson Nano and Jetson Xavier NX modules included as part of the Jetson Nano developer kit and the Jetson Xavier NX developer kit have slots for using microSD cards instead of eMMC as system storage devices.
So, my question is: did you build a custom setup, or what version of NX are you using? because it seems that the NX dev kit only has support for SD card instead of eMMC.

kmilo7204, 2020-08-04:
Hi @acostach and @JohnBetaCode 

It was our issue as we were not aware of the different versions:

As soon as we get the net NX Module we will give a try.

[image]

I'll close this issue as this definitely solves the problem. Hope it helps other people in the Balena community.

Cheers.

closed 2020-08-04 by kmilo7204

acostach, 2020-08-04:
Hi @kmilo7204 thanks for sharing this info. Indeed, the B01 carrier board + the NX module with eMMC was the only supported combo that was available for order at that time. The devkit was launched later, but plugging the eMMC version in it should do.

dremsol, 2020-10-13:
@klvnptr, this is not (yet) supported. However, it should not be that difficult contributing this machine by PR based on the emmc version.

klvnptr, 2020-10-13:
Thanks @dremsol . Do I need to only create a PR for jetson-flash, or I need to make a brand new balenos image as well? Thanks

dremsol, 2020-10-14:
A PR will be needed for both  `jetson-flash` and `balena-jetson` indeed.

acostach, 2020-11-02:
@JohnBetaCode @klvnptr @dremsol @kmilo7204 support for Jetson Xavier NX Devkit SD-CARD is now available for testing at https://dashboard.balena-staging.com/

You can download and unpack the image from there, write it to an sd-card and then plug the card in the devkit.

klvnptr, 2020-11-02:
Thank you @acostach . Is this with host extensions?

acostach, 2020-11-02:
@klvnptr no, that part is work in progress