balena-os/jetson-flash #29: Jetson Nano eMMC flash stalling

spiderkeys, 2020-09-08:
Board Model: Nano Devkit B01 (eMMC version, P3450)
OS version: v2.56.0+rev1
OS edition: Development

I've tried a few times now, following the instructions as written in the readme, but I keep getting stuck at the same point in the flashing process, shown in the logs below. Any ideas what the issue might be?

Output:
```
~/jetson-flash$ sudo ./bin/cmd.js -f ~/Downloads/balena-nano-emmc.img -m jetson-nano-emmc -p -o ./output
Consistency check done for /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/Linux_for_Tegra
Saved in /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts
Successfully generated hash
Checking resin cache
Cache image check done
Extracting partition LNX from /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin/img to /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin
Extracting partition DTB from /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin/img to /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin
Extracting partition RP1 from /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin/img to /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin
Extracting partition BMP from /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin/img to /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin
Extracting partition resin-boot from /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin/img to /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin
Extracting partition resin-rootA from /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin/img to /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin
Extracting partition resin-rootB from /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin/img to /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin
Extracting partition resin-state from /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin/img to /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin
Extracting partition resin-data from /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin/img to /home/spiderkeys/jetson-flash/output/jetson-flash-artifacts/resin
Successfully generated hash
###############################################################################
# L4T BSP Information:
# R32 , REVISION: 4.2
###############################################################################
# Target Board Information:
# Name: jetson-nano-emmc, Board Family: t210ref, SoC: Tegra 210, 
# OpMode: production, Boot Authentication: , 
###############################################################################
./tegraflash.py --chip 0x21 --applet "/tmp/Linux_for_Tegra/bootloader/nvtboot_recovery.bin" --skipuid --cmd "dump eeprom boardinfo cvm.bin" 
/tmp/Linux_for_Tegra/bootloader/tegraflash_internal.py:1860: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  if sig_type is not "zerosbk":
/tmp/Linux_for_Tegra/bootloader/tegraflash_internal.py:1863: SyntaxWarning: "is" with a literal. Did you mean "=="?
  if sig_type is "oem-rsa":
/tmp/Linux_for_Tegra/bootloader/tegraflash_internal.py:2294: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  while count is not 0 and not check_ismb2():
/tmp/Linux_for_Tegra/bootloader/tegraflash_internal.py:2303: SyntaxWarning: "is not" with a literal. Did you mean "!="?
  while count is not 0:
Welcome to Tegra Flash
version 1.0.0
Type ? or help for help and q or quit to exit
Use ! to execute system commands
 
[   0.0070 ] Generating RCM messages
[   0.0096 ] tegrarcm --listrcm rcm_list.xml --chip 0x21 0 --download rcm /tmp/Linux_for_Tegra/bootloader/nvtboot_recovery.bin 0 0
[   0.0123 ] RCM 0 is saved as rcm_0.rcm
[   0.0140 ] RCM 1 is saved as rcm_1.rcm
[   0.0140 ] List of rcm files are saved in rcm_list.xml
[   0.0141 ] 
[   0.0141 ] Signing RCM messages
[   0.0171 ] tegrasign --key None --list rcm_list.xml --pubkeyhash pub_key.key
[   0.0197 ] Assuming zero filled SBK key
[   0.0391 ] 
[   0.0392 ] Copying signature to RCM mesages
[   0.0422 ] tegrarcm --chip 0x21 0 --updatesig rcm_list_signed.xml
[   0.0659 ] 
[   0.0661 ] Boot Rom communication
[   0.0692 ] tegrarcm --chip 0x21 0 --rcm rcm_list_signed.xml --skipuid
[   0.0722 ] RCM version 0X210001
[   0.3486 ] Boot Rom communication completed
[   1.3560 ] 
[   1.3562 ] dump EEPROM info
[   1.3594 ] tegrarcm --oem platformdetails eeprom /tmp/Linux_for_Tegra/bootloader/cvm.bin
[   1.3619 ] Applet version 00.01.0000
[   1.6103 ] Saved platform info in /tmp/Linux_for_Tegra/bootloader/cvm.bin
[   1.7149 ] 
Board ID(3448) version(400) 
[... 131 lines cut ...]
[   1.1774 ] 
[   1.1784 ] tegrasign --key None --list tegra210-p3448-0000-p3449-0000-b00.dtb_list.xml
[   1.1791 ] Assuming zero filled SBK key
[   1.1843 ] 
[   1.1855 ] tegrahost --updatesigheader tegra210-p3448-0000-p3449-0000-b00.dtb_blheader.encrypt tegra210-p3448-0000-p3449-0000-b00.dtb_blheader.hash zerosbk
[   1.1866 ] 
[   1.1869 ] Sending bootloader and pre-requisite binaries
[   1.1876 ] tegrarcm --download ebt cboot.bin.encrypt 0 0 --download rp1 tegra210-p3448-0000-p3449-0000-b00.dtb.encrypt 0
[   1.1883 ] Applet version 00.01.0000
[   1.4515 ] Sending ebt
[   1.4521 ] [................................................] 100%
[   1.5586 ] Sending rp1
[   1.5627 ] [................................................] 100%
[   1.6140 ] 
[   1.6172 ] tegrarcm --boot recovery
[   1.6196 ] Applet version 00.01.0000
[   1.8718 ] 
[   1.8720 ] Retrieving storage infomation
[   1.8749 ] tegrarcm --oem platformdetails storage storage_info.bin
[   1.8776 ] Applet is not running on device. Continue with Bootloader
[   2.4219 ] 
[   2.4264 ] tegradevflash --oem platformdetails storage storage_info.bin
[   2.4288 ] Cboot version 00.01.0000
[   2.5460 ] Saved platform info in storage_info.bin
```

spiderkeys, 2020-09-08:
In case it is useful, here is the output upon sending sigint:
```
^CTraceback (most recent call last):
  File "/tmp/Linux_for_Tegra/bootloader/tegraflash_internal.py", line 2235, in tegraflash_get_storage_info
    run_command(command)
  File "/tmp/Linux_for_Tegra/bootloader/tegraflash_internal.py", line 199, in run_command
    raise tegraflash_exception('Return value ' + str(return_code) +
tegraflash_internal.tegraflash_exception: 'Return value 8\nCommand tegrarcm --oem platformdetails storage storage_info.bin'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./tegraflash.py", line 1280, in <module>
    tegraflash_run_commands()
  File "./tegraflash.py", line 1149, in tegraflash_run_commands
    interpreter.onecmd(command)
  File "/usr/lib/python3.8/cmd.py", line 217, in onecmd
    return func(arg)
  File "./tegraflash.py", line 209, in do_flash
    tegraflash_flash(exports)
  File "/tmp/Linux_for_Tegra/bootloader/tegraflash_internal.py", line 266, in tegraflash_flash
    tegraflash_get_storage_info()
  File "/tmp/Linux_for_Tegra/bootloader/tegraflash_internal.py", line 2239, in tegraflash_get_storage_info
    run_command(command)
  File "/tmp/Linux_for_Tegra/bootloader/tegraflash_internal.py", line 195, in run_command
    log = print_process(process, enable_print)
  File "/tmp/Linux_for_Tegra/bootloader/tegraflash_internal.py", line 155, in print_process
    output = process.stdout.read(1)
KeyboardInterrupt
```

acostach, 2020-09-08:
Hi, if you download and extract the BSP from here: https://developer.nvidia.com/embedded/linux-tegra-r32.4.2 and flash the board, does it flash fine?

For instance:
```
tar xf Tegra210_Linux_R32.4.2_aarch64.tbz2
cd Linux_for_Tegra
sudo ./flash.sh jetson-nano-emmc mmcblk0p1
```

spiderkeys, 2020-09-08:
Just tried your suggestion, and it got stuck at the same point.

Another user on the Nvidia forums describes an identical problem, with the difference being that they are using the A02 dev-kit.
https://forums.developer.nvidia.com/t/flashing-jetson-nono-emmc/82103

I've confirmed that both the dev-kit carrier board and the module itself are labeled as the B01 variant, which should support the eMMC.

acostach, 2020-09-09:
jetson-flash uses flash.sh from nvidia's BSP package for writing the image on the board, and in this case it looks like flashing doesn't work with the BSP either, at least not for the emmc machine.

Does it do the same if using flash.sh with 'jetson-nano-qspi' and 'jetson-nano-qspi-sd' ?

spiderkeys, 2020-09-09:
Flashing with "jetson-nano-qspi" via the BSP script was successful:
```
~/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra$ sudo ./flash.sh jetson-nano-qspi mmcblk0p1
###############################################################################
# L4T BSP Information:
# R32 , REVISION: 4.3
###############################################################################
# Target Board Information:
# Name: jetson-nano-qspi, Board Family: t210ref, SoC: Tegra 210, 
# OpMode: production, Boot Authentication: , 
###############################################################################
./tegraflash.py --chip 0x21 --applet "/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/nvtboot_recovery.bin" --skipuid --cmd "dump eeprom boardinfo cvm.bin" 
Welcome to Tegra Flash
version 1.0.0
Type ? or help for help and q or quit to exit
Use ! to execute system commands
 
[   0.0014 ] Generating RCM messages
[   0.0020 ] tegrarcm --listrcm rcm_list.xml --chip 0x21 0 --download rcm /home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/nvtboot_recovery.bin 0 0
[   0.0026 ] RCM 0 is saved as rcm_0.rcm
[   0.0047 ] RCM 1 is saved as rcm_1.rcm
[   0.0047 ] List of rcm files are saved in rcm_list.xml
[   0.0047 ] 
[   0.0047 ] Signing RCM messages
[   0.0071 ] tegrasign --key None --list rcm_list.xml --pubkeyhash pub_key.key
[   0.0076 ] Assuming zero filled SBK key
[   0.0211 ] 
[   0.0211 ] Copying signature to RCM mesages
[   0.0218 ] tegrarcm --chip 0x21 0 --updatesig rcm_list_signed.xml
[   0.0226 ] 
[   0.0227 ] Boot Rom communication
[   0.0232 ] tegrarcm --chip 0x21 0 --rcm rcm_list_signed.xml --skipuid
[   0.0237 ] RCM version 0X210001
[   0.2733 ] Boot Rom communication completed
[   1.2806 ] 
[   1.2808 ] dump EEPROM info
[   1.2858 ] tegrarcm --oem platformdetails eeprom /home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/cvm.bin
[   1.2894 ] Applet version 00.01.0000
[   1.4998 ] Saved platform info in /home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/cvm.bin
[   1.5752 ] 
Board ID(3448) version(400) 
copying bctfile(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/t210ref/BCT/P3448_A00_4GB_Micron_4GB_lpddr4_204Mhz_P987.cfg)... done.
copying bootloader(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/t210ref/cboot.bin)... done.
copying initrd(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/l4t_initrd.img)... done.
	populating kernel to rootfs... done.
	populating initrd to rootfs... done.
	populating /home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/kernel/dtb/tegra210-p3448-0000-p3449-0000-b00.dtb to rootfs... done.
Making Boot image... done.
Existing sosfile(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/nvtboot_recovery.bin) reused.
copying tegraboot(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/t210ref/nvtboot.bin)... done.
copying cpu_bootloader(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/t210ref/cboot.bin)... done.
copying bpffile(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/t210ref/sc7entry-firmware.bin)... done.
Existing badpagefile(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/badpage.bin) reused.
copying wb0boot(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/t210ref/warmboot.bin)... done.
Existing tosfile(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/tos-mon-only.img) reused.
Existing eksfile(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/eks.img) reused.
copying dtbfile(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/kernel/dtb/tegra210-p3448-0000-p3449-0000-b00.dtb)... done.
Copying nv_boot_control.conf to rootfs
Making system.img... 
	populating rootfs from /home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/rootfs ... done.
	populating /boot/extlinux/extlinux.conf ... done.
	Sync'ing system.img ... done.
	Converting RAW image to Sparse image... done.
system.img built successfully. 
Existing tbcfile(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/nvtboot_cpu.bin) reused.
copying tbcdtbfile(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/kernel/dtb/tegra210-p3448-0000-p3449-0000-b00.dtb)... done.
copying cfgfile(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/t210ref/cfg/flash_l4t_t210_spi_p3448.xml) to flash.xml... done.
copying flasher(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/t210ref/cboot.bin)... done.
Existing flashapp(/home/spiderkeys/Downloads/Tegra210_Linux_R32.4.3_aarch64/Linux_for_Tegra/bootloader/tegraflash.py) reused.
./tegraflash.py --bl cboot.bin --bct  P3448_A00_4GB_Micron_4GB_lpddr4_204Mhz_P987.cfg --odmdata 0x94000 --bldtb tegra210-p3448-0000-p3449-0000-b00.dtb --applet nvtboot_recovery.bin --cmd "flash; reboot"  --cfg flash.xml --chip 0x21    --bins "EBT cboot.bin; DTB tegra210-p3448-0000-p3449-0000-b00.dtb" 
[... 405 lines cut ...]
[  39.8743 ] [................................................] 100%
[... cut ...]
[ 183.2406 ] Coldbooting the device
[ 183.2434 ] tegradevflash --reboot coldboot
[ 183.2459 ] Cboot version 00.01.0000
[ 183.4172 ] 
*** The target t210ref has been flashed successfully. ***
Reset the board to boot from internal eMMC.
```
Despite this being successful, the system does not boot from the SD card used during the flash. Switching back to an etcher-flashed balena SD card after that does boot correctly.

I tried using "jetson-nano-emmc" again after all of the above, but it still hangs.

spiderkeys, 2020-09-09:
Digging into this a bit further, I think this may just be a situation where the part numbering for the module is highly confusing, and there actually is no eMMC available on this module.

To help clear things up for anyone else that runs into this, my module part number on the PCB reads:
180-13448-DAAA-B01

But according to these posts, it seems that this PCB part number is unrelated to whether or not eMMC is available:
https://forums.developer.nvidia.com/t/jetson-nano-b01-no-emmc/116202
https://forums.developer.nvidia.com/t/confusing-in-board-name-for-the-jetson-nano-developer-kit/79547

closed 2020-09-09 by spiderkeys

acostach, 2020-09-09:
Thank you for the update, great to hear it worked fine