balena-os/jetson-flash #32: ./bin/cmd.js -f balena.img -m jetson-nano-qspi-sd faild

lminggang, 2020-11-13:
$node --trace-warnings ./bin/cmd.js -f balena.img -m jetson-nano-qspi-sd

Consistency check done for /tmp/23387/Linux_for_Tegra
Saved in /tmp/23387
Successfully generated hash
Checking resin cache
Cache image check done
(node:23387) UnhandledPromiseRejectionWarning: SyntaxError: Invalid MBR boot signature. Expected 0xAA55, but saw 0x0
    at Function.MBR.detectFormat (/home/parallels/Downloads/jetson-flash/node_modules/mbr/lib/mbr.js:73:11)
    at Function.MBR.parse (/home/parallels/Downloads/jetson-flash/node_modules/mbr/lib/mbr.js:54:20)
    at new MBR (/home/parallels/Downloads/jetson-flash/node_modules/mbr/lib/mbr.js:12:16)
    at getPartitionsFromMBRBuf (/home/parallels/Downloads/jetson-flash/node_modules/partitioninfo/build/partitioninfo.js:49:11)
    at /home/parallels/Downloads/jetson-flash/node_modules/partitioninfo/build/partitioninfo.js:100:18
    at emitUnhandledRejectionWarning (internal/process/promises.js:168:15)
    at processPromiseRejections (internal/process/promises.js:247:11)
    at processTicksAndRejections (internal/process/task_queues.js:94:32)
(node:23387) SyntaxError: Invalid MBR boot signature. Expected 0xAA55, but saw 0x0
    at Function.MBR.detectFormat (/home/parallels/Downloads/jetson-flash/node_modules/mbr/lib/mbr.js:73:11)
    at Function.MBR.parse (/home/parallels/Downloads/jetson-flash/node_modules/mbr/lib/mbr.js:54:20)
    at new MBR (/home/parallels/Downloads/jetson-flash/node_modules/mbr/lib/mbr.js:12:16)
    at getPartitionsFromMBRBuf (/home/parallels/Downloads/jetson-flash/node_modules/partitioninfo/build/partitioninfo.js:49:11)
    at /home/parallels/Downloads/jetson-flash/node_modules/partitioninfo/build/partitioninfo.js:100:18
(node:23387) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
    at emitDeprecationWarning (internal/process/promises.js:180:11)
    at processPromiseRejections (internal/process/promises.js:249:13)
    at processTicksAndRejections (internal/process/task_queues.js:94:32)

lminggang, 2020-11-13:
Step:
1. git clone https://github.com/balena-os/jetson-flash.git
2. install node-v14.15.0-linux-x64
3. cd jetson-flash
4. npm install
5. torch balena.img
6. npm install --save file-disk@7.0.1
7. node --trace-warnings ./bin/cmd.js -f balena.img -m jetson-nano-qspi-sd

acostach, 2020-11-13:
What does `parted -s balena.img unit s print` show? If you have the qspi-sd you can flash the image to the sd-card on your PC and then plug it in the device, it's faster.

acostach, 2022-01-25:
No reply received so I'll close this for now, feel free to re-open if needed.

closed 2022-01-25 by acostach