balena-os/jetson-flash #37: Error: Cannot find module 'yargs' : with jetson-flash with nano

venkatbhats, 2021-02-09:
Hi I tried to flash my nvidia Jetson nano emmc and getting below error, please help:

hm@hm:~/BalenoOS$ ls
balena-cloud-jetson-nano-emmc-2.67.3+rev2-dev-v12.3.0.img      jetson-flash  package-lock.json
balena-cloud-jetson-nano-emmc-2.67.3+rev2-dev-v12.3.0.img.zip  node_modules
hm@hm:~/BalenoOS$ sudo su
[sudo] password for hm:
root@hm:/home/hm/BalenoOS# ./jetson-flash/bin/cmd.js -f balena-cloud-jetson-nano-emmc-2.67.3+rev2-dev-v12.3.0.img -m jetson-nano-emmc
internal/modules/cjs/loader.js:883
  throw err;
  ^

Error: Cannot find module 'yargs'
Require stack:
- /home/hm/BalenoOS/jetson-flash/bin/cmd.js
    at Function.Module._resolveFilename (internal/modules/cjs/loader.js:880:15)
    at Function.Module._load (internal/modules/cjs/loader.js:725:27)
    at Module.require (internal/modules/cjs/loader.js:952:19)
    at require (internal/modules/cjs/helpers.js:88:18)
    at Object.<anonymous> (/home/hm/BalenoOS/jetson-flash/bin/cmd.js:23:15)
    at Module._compile (internal/modules/cjs/loader.js:1063:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1092:10)
    at Module.load (internal/modules/cjs/loader.js:928:32)
    at Function.Module._load (internal/modules/cjs/loader.js:769:14)
    at Function.executeUserEntryPoint [as runMain] (internal/modules/run_main.js:72:12)
    at internal/main/run_main_module.js:17:47 {
  code: 'MODULE_NOT_FOUND',
  requireStack: [ '/home/hm/BalenoOS/jetson-flash/bin/cmd.js' ]
}

venkatbhats, 2021-02-09:
I could resolve above error by installing dependency, but now I am facing below issue:
Consistency check done for /tmp/11064/Linux_for_Tegra
\ Downloading and extracting https://developer.nvidia.com/embedded/L4T/r32_Release_v4.4/r32_Release_v4.4-GMC3/T210/Tegra210_Linux_R32.4.4_aarch64.tbz2internal/streams/legacy.js:61
      throw er; // Unhandled stream error in pipe.
      ^

Error: unable to get local issuer certificate
    at TLSSocket.onConnectSecure (_tls_wrap.js:1497:34)
    at TLSSocket.emit (events.js:315:20)
    at TLSSocket._finishInit (_tls_wrap.js:932:8)
    at TLSWrap.ssl.onhandshakedone (_tls_wrap.js:706:12) {
  code: 'UNABLE_TO_GET_ISSUER_CERT_LOCALLY'
}

acostach, 2021-04-07:
Hi @venkatbhats, the dependency list for running this tool is visible in https://github.com/balena-os/jetson-flash/blob/a166a5cf04563c55055b413e24af9a7f6746c5e3/package.json#L24
so you can install them by issuing:
```
cd jetson-flash
npm install yargs file-disk ...
```

Second issue looks like a certificates problem, in order to speed-up flashing and not wait for the BSP archive to download from the internet you can setup a local http server on your PC, download the BSP archive to it's http folder and switch this line:
https://github.com/balena-os/jetson-flash/blob/a166a5cf04563c55055b413e24af9a7f6746c5e3/lib/resin-jetson-flash.js#L64
to 
`http://127.0.0.1/Tegra210_Linux_R32.4.4_aarch64.tbz2`
...

acostach, 2022-01-25:
No feedback received, closing for now. Feel free to re-open.

closed 2022-01-25 by acostach