balena-os/jetson-flash #38: BalenaOS on nvidia Jetson Nano devkit not booting to OS -- emmc image

venkatbhats, 2021-02-10:
Hi,

I am trying balenaOS on NVIDIA Jetson Nano Devkit and using emmc Image to boot on nvidia based customized carrier board.

 the module is not getting booted up, It hangs with constantly displaying balenaOS Logo on screen,
I am unable to see serial console log as well. but  my USB mouse and keyboards are getting detected as I can see num+lock and caps+lock keys are accessible.

used below image to flash emmc  from linux host: balena-cloud-jetson-nano-emmc-2.67.3+rev2-v12.3.0.img.zip

steps:
Installed all the dependencies
using Linux based host - Ubuntu 18
Run the cli, specifying desired device type: device type Jetson Nano emmc
flash command:
root@hx:/home/hx/BalenoOS# ./jetson-flash/bin/cmd.js -f balena-cloud-jetson-nano-emmc-2.67.3+rev2-dev-v12.3.0.img -m jetson-nano-emmc

followed below instructions:
https://github.com/balena-os/jetson-flash

Please help me on this.

acostach, 2021-02-22:
Hi, if the board displays the BalenaOS logo it means that has booted correctly. The image does not come with a graphical interface, but one can be installed in the application containers. If you connect the board to the internet using an ethernet cable, the device should be visible in your application dashboard.

venkatbhats, 2021-02-25:
Hi, Thanks for your help. I later understood that. Will get back to you if need further help.

acostach, 2021-02-25:
Ok, feel free to re-open the ticket if necessary.

closed 2021-02-25 by acostach