balena-os/jetson-flash #39: jetson: investigate flashing preloaded images

acostach, 2021-02-17:
The xml size has a default value for the resin-data partition, which is smaller that the size of the actual data partition if the balenaOS image has an app preloaded.

AeroNotix, 2021-03-14:
@acostach I believe I am running into this issue. Is there a workaround or fix/

acostach, 2021-03-15:
Hi @AeroNotix ,

You can use parted on your image to find out the data partition size in bytes, and then increase this size here:https://github.com/balena-os/jetson-flash/blob/a69a690f73ae502606071dc7bb8e2f20d98cf30f/assets/jetson-tx2-assets/resinOS-flash.xml#L428 to that value or a few bytes larger.

AeroNotix, 2021-03-15:
@acostach Should I correct all of them? None of them look correct in the XML file vs what I see for the partition sizes on a preloaded image:

```
Number  Start   End     Size    File system  Name         Flags
 1      4194kB  88.1MB  83.9MB  fat16        resin-boot   boot, esp
 2      88.1MB  14.5GB  14.4GB  ext4         resin-rootA
 3      14.5GB  14.5GB  4194kB               resin-rootB
 4      14.5GB  14.5GB  4194kB               resin-state
 5      14.5GB  14.5GB  15.7MB               resin-data
```

acostach, 2021-03-16:
@AeroNotix no, you should only try change the one for the data partition, but use size in bytes from the actual emmc image:
```
        <partition name="resin-data" type="data">
            <allocation_policy> sequential </allocation_policy>
            <filesystem_type> basic </filesystem_type>
            <size> 1085276160 </size> <-------- this one
            <file_system_attribute> 0 </file_system_attribute>
```

So the TX2 image from the dashboard is a flasher image type, which can be booted from the sd-card and only has 2 filesystems (boot and rootA). Inside rootA of this flasher image, at /opt/resin-image-jetson-tx2.resinos-img, is the actual image that gets written to the device's emmc. 

```
$ parted -s resin-image-jetson-tx2.resinos-img unit B print
Number  Start        End          Size        File system  Name              Flags
 1      2097664B     6291967B     4194304B                 mts-bootpack
 2      6291968B     10486271B    4194304B                 mts-bootpack_b
 3      10486272B    11010559B    524288B                  cpu-bootloader
 4      11010560B    11534847B    524288B                  cpu-bootloader_b
 5      11534848B    12059135B    524288B                  bootloader-dtb
 6      12059136B    12583423B    524288B                  bootloader-dtb_b
 7      12583424B    15729151B    3145728B                 secure-os
 8      15729152B    18874879B    3145728B                 secure-os_b
 9      18874880B    19923455B    1048576B                 eks
10      19923456B    20972031B    1048576B                 adsp-fw
11      20972032B    22020607B    1048576B                 adsp-fw_b
12      22020608B    22639103B    618496B                  bpmp-fw
13      22639104B    23257599B    618496B                  bpmp-fw_b
14      23257600B    24281599B    1024000B                 bpmp-fw-dtb
15      24281600B    25305599B    1024000B                 bpmp-fw-dtb_b
16      25305600B    26354175B    1048576B                 sce-fw
17      26354176B    27402751B    1048576B                 sce-fw_b
18      27402752B    29499903B    2097152B                 sc7
19      29499904B    31597055B    2097152B                 sc7_b
20      31597056B    33694207B    2097152B                 kernel
21      33694208B    35791359B    2097152B                 kernel_b
22      35791360B    36315647B    524288B                  kernel-dtb
23      36315648B    36839935B    524288B                  kernel-dtb_b
24      50331648B    134217727B   83886080B   fat16        resin-boot        boot, esp
25      134217728B   633339903B   499122176B  ext4         resin-rootA
26      633339904B   1132462079B  499122176B  ext4         resin-rootB
27      1132462080B  1153433599B  20971520B   ext4         resin-state
28      1153433600B  1720324607B  566891008B  ext4         resin-data
```

In this example the value in bytes for the preloaded data partition is  566891008

acostach, 2021-03-16:
@AeroNotix I've merged https://github.com/balena-os/jetson-flash/pull/42 so this should work by default now in v0.5.7, without editing any xml. Feel free to re-open if necessary.

closed 2021-03-16 by acostach

AeroNotix, 2021-03-16:
Awesome! Looks like you can close #8 as well maybe?

AeroNotix, 2021-03-16:
Good work Sir, always a pleasure seeing a maestro at work.