balena-os/jetson-flash #43: Can't seem to flash images built from 2.73.1

AeroNotix, 2021-03-29:
Hi, 

I built an image with the latest commit of https://github.com/balena-os/balena-jetson

when I go to flash it with jetson-flash I receive:

```
(node:20) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'name' of undefined
    at Object.exports.extractPartition (/jetson-flash/lib/utils.js:137:38)
(node:20) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:20) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

When I use a 2.73.1 image downloaded from the dashboard (filename: balena-cloud-TestMFP-jetson-tx2-2.73.1+rev1-dev-v12.4.6.img) I get the same error.

acostach, 2021-03-29:
Hello Aaron, thanks for letting us know. I'll have a look into this and get back to you

acostach, 2021-03-30:
Hi Aaron, this issue is fixed in v0.5.8. Please re-open this ticket if the error persists on your side.

closed 2021-03-30 by acostach