balena-os/jetson-flash #48: Fatal error on retreiving balena os image when trying to flash jetson tx2 nx module

eli-davis, 2021-05-03:
Hi,

I am trying to flash a jetson tx2 nx module using the balena os image from their website. Below is the error readout:

Successfully generated hash
Checking resin cache
Cache image check done
Retrieve BalenaOS image from balena-cloud-jetson-tx2-2.73.1+rev4-v12.4.6.img

#
# Fatal error in , line 0
# Check failed: result.second.
#
#
#
#FailureMessage Object: 0x7ffd8ddb0770
 1: 0xb7db41  [node]
 2: 0x1c15474 V8_Fatal(char const*, ...) [node]
 3: 0x100c201 v8::internal::GlobalBackingStoreRegistry::Register(std::shared_ptr<v8::internal::BackingStore>) [node]
 4: 0xd23818 v8::ArrayBuffer::GetBackingStore() [node]
 5: 0xaea77f node::Buffer::Data(v8::Local<v8::Object>) [node]
 6: 0xb215ff  [node]
 7: 0xd5f70b  [node]
 8: 0xd60bac  [node]
 9: 0xd61226 v8::internal::Builtin_HandleApiCall(int, unsigned long*, v8::internal::Isolate*) [node]
10: 0x160c579  [node]
Illegal instruction (core dumped)

Am I doing something wrong?

acostach, 2021-05-05:
Hi @eli-davis , the TX2 image from the dashboard is for the Jetson TX2 module, not for the Jetson TX2 NX.

Regarding the flashing error, what Node js version do you use? Does it work for you if you revert to jetson-flash v0.5.8?

acostach, 2021-05-05:
Ok, looks like it's an issue in one of the node modules and it's related only to the node version - starts happening with node 14, no matter the version of jetson-flash used. Not trivial to trace and debug unfortunately.

For now please use nodejs v10 or v12.

acostach, 2023-03-27:
Fixed by https://github.com/balena-os/jetson-flash/pull/106

closed 2023-03-27 by acostach