balena-os/jetson-flash #55: rror: Cannot copy 'p3509-0000+p3668-0001-qspi-emmc.conf' to a subdirectory of itself, 'p3509-0000+p3668-0001-qspi-emmc.conf'.

minusplusminus, 2021-11-15:
``
sudo ./bin/cmd.js  -f "./balena-cloud-neonlink-nano-dist-jetson-xavier-nx-devkit-emmc-2.82.11+rev8-dev-v12.9.3.img" -m jetson-xavier-nx-devkit-emmc 
Consistency check done for /tmp/107072/Linux_for_Tegra
Saved in /tmp/107072
Successfully generated hash
Checking resin cache
Cache image check done
Extracting partition kernel from /tmp/107072/resin/img to /tmp/107072/resin
Extracting partition kernel_b from /tmp/107072/resin/img to /tmp/107072/resin
Extracting partition kernel-dtb from /tmp/107072/resin/img to /tmp/107072/resin
Extracting partition kernel-dtb_b from /tmp/107072/resin/img to /tmp/107072/resin
Extracting partition resin-boot from /tmp/107072/resin/img to /tmp/107072/resin
Extracting partition resin-rootA from /tmp/107072/resin/img to /tmp/107072/resin
Extracting partition resin-rootB from /tmp/107072/resin/img to /tmp/107072/resin
Extracting partition resin-state from /tmp/107072/resin/img to /tmp/107072/resin
Extracting partition resin-data from /tmp/107072/resin/img to /tmp/107072/resin
Data partition size in bytes: 212860928
Successfully generated hash
(node:107072) UnhandledPromiseRejectionWarning: Error: Cannot copy 'p3509-0000+p3668-0001-qspi-emmc.conf' to a subdirectory of itself, 'p3509-0000+p3668-0001-qspi-emmc.conf'.
    at /home/parallels/jetson-flash/node_modules/fs-extra/lib/copy/copy.js:210:21
    at FSReqCallback.oncomplete (fs.js:156:23)
(node:107072) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). To terminate the node process on unhandled promise rejection, use the CLI flag `--unhandled-rejections=strict` (see https://nodejs.org/api/cli.html#cli_unhandled_rejections_mode). (rejection id: 1)
(node:107072) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.

``

minusplusminus, 2021-11-16:
Solved it by removing /etc/Linux_for_Tegra directory

acostach, 2022-01-25:
Closing as per the above feedback.

closed 2022-01-25 by acostach

msmcconnell, 2024-04-25:
> Solved it by removing /etc/Linux_for_Tegra directory

@minusplusminus I'm having the same issue. Is your fix to remove that folder from the NX device itself?