balena-os/jetson-flash #69: Unhandled stream error in pipe during flashing Balena Image on Jetson Nano eMMc

AmirAvonic, 2022-04-01:
We are encountering a problem all of a sudden during the flashing of a Jetson Nano eMMc with an image that results in the following error: 

`internal/streams/legacy.js:57
      throw er; // Unhandled stream error in pipe.
      ^
Error
    at new Bzip2Error (/jetson-flash/node_modules/unbzip2-stream/lib/bzip2.js:31:19)
    at Object.Error (/jetson-flash/node_modules/unbzip2-stream/lib/bzip2.js:36:37)
    at Object.bzip2.header (/jetson-flash/node_modules/unbzip2-stream/lib/bzip2.js:152:39)
    at decompressBlock (/jetson-flash/node_modules/unbzip2-stream/index.js:18:29)
    at decompressAndQueue (/jetson-flash/node_modules/unbzip2-stream/index.js:48:20)
    at Stream.end (/jetson-flash/node_modules/unbzip2-stream/index.js:83:17)
    at _end (/jetson-flash/node_modules/through/index.js:65:9)
    at Stream.stream.end (/jetson-flash/node_modules/through/index.js:74:5)
    at Request.onend (internal/streams/legacy.js:42:10)
    at Request.emit (events.js:203:15)
    at IncomingMessage.<anonymous> (/home/amir-avonic/jetson-flash/node_modules/request/request.js:1076:12)
    at Object.onceWrapper (events.js:286:20)
    at IncomingMessage.emit (events.js:203:15)
    at endReadableNT (_stream_readable.js:1145:12)
    at process._tickCallback (internal/process/next_tick.js:63:19)
`

I am using the latest version of the Jetson-Flash repository, NPM version: 6.14.12 and Nodejs version: v10.24.1. The BalenaOS version of the image is 2.67.3+rev2.  The error appears during the process of downloading and extracting the L4T binaries:

`Downloading and extracting https://developer.nvidia.com/embedded/L4T/r32_Release_v4.4/r32_Release_v4.4-GMC3/T210/Tegra210_Linux_R32.4.4_aarch64.tbz2`

Any idea where this problem could suddenly be coming from?

acostach, 2022-04-01:
Hi @AmirAvonic , it might be a problem with the archive that's being downloaded.

Could you check if this still happens if you set up a local http server to provide the already downloaded image to Jetson Flash?

A simple python server that's already available on the internet would look like this:

```
cat simplehttp 
#!/usr/bin/python
import SimpleHTTPServer
import SocketServer

PORT = 80

Handler = SimpleHTTPServer.SimpleHTTPRequestHandler

httpd = SocketServer.TCPServer(("", PORT), Handler)

print "serving at port", PORT
httpd.serve_forever()

```

and you will need to manually download the BSP archive and place it in the same directory as simplehttp, which will thn be run with ./simplehttp.

lib/resin-jetson-flash.js will need to be modified to point to localhost:

```
['jetson-nano-emmc']: {
                                url:
                                        'http://127.0.0.1:/Tegra210_Linux_R32.4.4_aarch64.tbz2',
                                partitions: {

```

AmirAvonic, 2022-04-01:
Ah yes, apparently the NVIDIA Archives are for a large part unreachable. Going to the weblink I provided resulted in a 404 error on the webpage, and the same applies to a bunch of other versions. The only version that is still available is 32.5 and your method indeed solved the problem! 

Thanks and also appreciate the swift reply!

closed 2022-04-01 by AmirAvonic