balena-os/jetson-flash #70: 'No magic number found' when flashing to devkit.

AlexandreLeboucher, 2022-04-01:
Followed the steps here without VM on Ubuntu 18. This worked perfectly for the past month but stopped working today.

https://forums.balena.io/t/jetson-nano-2gb-not-showing-up-on-dashboard/350033/10

Any idea what may be causing it ?

Here's the trace :

(base) alex@alex-XPS-15-9560:~/Vizzia/jetson-flash$ sudo ./bin/cmd.js -f /home/alex/Vizzia/jetson-flash/balena-cloud-VizziaDev2-jetson-nano-2.88.4+rev1-dev-v12.11.0.img -m jetson-nano-qspi-sd
Consistency check done for /tmp/15210/Linux_for_Tegra
| Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6./ Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.- Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.\ Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.| Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6./ Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.- Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.\ Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.| Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6./ Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.- Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.\ Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.| Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6./ Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.- Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.\ Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.| Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6./ Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.- Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.\ Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.| Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6./ Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.- Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.\ Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.| Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6./ Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.- Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.\ Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.| Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6./ Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.- Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.\ Downloading and extracting https://developer.nvidia.com/embedded/l4t/r32_release_v6.1/t210/jetson-210_linux_r32.6.1_aarch64.tbz2
internal/streams/legacy.js:61
      throw er; // Unhandled stream error in pipe.
      ^
Error
    at new Bzip2Error (/home/alex/Vizzia/jetson-flash/node_modules/unbzip2-stream/lib/bzip2.js:31:19)
    at Object.Error (/home/alex/Vizzia/jetson-flash/node_modules/unbzip2-stream/lib/bzip2.js:36:37)
    at Object.bzip2.header (/home/alex/Vizzia/jetson-flash/node_modules/unbzip2-stream/lib/bzip2.js:152:39)
    at decompressBlock (/home/alex/Vizzia/jetson-flash/node_modules/unbzip2-stream/index.js:18:29)
    at decompressAndQueue (/home/alex/Vizzia/jetson-flash/node_modules/unbzip2-stream/index.js:48:20)
    at Stream.end (/home/alex/Vizzia/jetson-flash/node_modules/unbzip2-stream/index.js:83:17)
    at _end (/home/alex/Vizzia/jetson-flash/node_modules/through/index.js:65:9)
    at Stream.stream.end (/home/alex/Vizzia/jetson-flash/node_modules/through/index.js:74:5)
    at Request.onend (internal/streams/legacy.js:46:10)
    at Request.emit (events.js:326:22)
    at IncomingMessage.<anonymous> (/home/alex/Vizzia/jetson-flash/node_modules/request/request.js:1076:12)
    at Object.onceWrapper (events.js:420:28)
    at IncomingMessage.emit (events.js:326:22)
    at endReadableNT (_stream_readable.js:1241:12)
    at processTicksAndRejections (internal/process/task_queues.js:84:21) {
  message: 'No magic number found'

squealteam6, 2022-04-02:
I think its because NVIDIA is having issues with their download server

acostach, 2022-04-02:
Hi @AlexandreLeboucher , Jetson Flash downloads the Nvidia BSP archive in order to use the tegra flashing tools. Looks like there has been a problem with the Nvidia servers but now they seem to be back online. If the problem occurs again, you can use this method of loading the BSP archive from a local web server: https://github.com/balena-os/jetson-flash/issues/69#issuecomment-1085622264

AlexandreLeboucher, 2022-04-02:
Great thanks for the support !