balena-os/jetson-flash #71: Faster way to re-flash only the L4T boot version fix?

msherman64, 2022-04-12:
I primarily use this tool to flash jetson-nanos with sdcard storage, in order to fix the issue where newer L4T (32.5.1+) versions screw up booting. Is there a way to only flash the "boot fix" portion, rather than the whole SDCard image? Flashing the whole image over USB is much slower than using a sdcard reader directly, where possible.

Thanks!

acostach, 2022-04-14:
Hi @msherman64 , this approach might work for you: you could try editing https://github.com/balena-os/jetson-flash/blob/master/assets/jetson-nano-qspi-sd-assets/resinOS-flash.xml#L207 to remove TBC, RP1, LNX etc and larger partitions like resin-boot, rootA-B, resin-state, resin-data and then flash the device using any sd-card that won't be used in production. After flashing is completed, you remove that sd-card and insert another one which has already been written on your PC with the Nano SD-CARD image from balena cloud.

acostach, 2022-06-07:
Closing for now, feel free to re-open if you have other questions.

closed 2022-06-07 by acostach