balena-os/jetson-flash #85: Support flashing to external NVME SSDs

devarshi, 2022-08-08:
In many commercial grade Jetson NX devices, there's support for NVME SSDs. It seems L4T 32.6.1 introduced support to boot from NVME SSDs (https://docs.nvidia.com/jetson/archives/r34.1/DeveloperGuide/text/SD/FlashingSupport.html#flashing-to-an-nvme-drive) . 

There's also a set of scripts here: https://github.com/jetsonhacks/bootFromExternalStorage/blob/main/flash_jetson_external_storage.sh to flash external storage. 

What would be a good starting point to bring those changes to this repo ?

nbraem, 2022-08-16:
Please note that the upcoming Jetson Orin NX has no builtin storage, so will always have to boot from an SSD. So this will be a must have for the Orin NX.

acostach, 2022-08-22:
Hi @devarshi @nbraem , balenaOS relies on having the OS partitions on the same medium as the tegra partitions. We currently do not support flashing and/or loading the rootfs from a different medium. We will let you know once and if this will change in the future. As an alternative, SSDs can be mounted in containers for storage expansion.

closed 2022-08-22 by acostach

devarshi, 2022-08-22:
Gotcha. Is there any reason why tegra partitions cannot also be on the SSD?

acostach, 2022-08-22:
@devarshi as far as I know all tegra partitions are on the eMMC and only the rootfs is stored on the alternate storage - SSD. This because the tegra firmware/bootloaders in the hw partition and/or spi flash cannot load the next elements in the boot chain from the SSD, this is not a surprise given that these bootloaders are very small. The link that you posted also offers instructions on writing the APP partition to the SSD only and no other partitions.