balena-os/jetson-flash #90: UnicodeDecodeError

etohana, 2022-10-20:
jetson-xavier-nx-devkit-emmc, Board Family: t186ref, SoC: Tegra 194, 
balena-cloud-parking-lot-jetson-xavier-nx-devkit-emmc-2.99.27+rev1-v14.0.8.img
Node.js 12.22.12

Greetings:
I am trying to flash BalenaOS on a Xavier NX Module (w/ eMMC). The procedure seems to proceed okay until this point:

> File "/tmp/Linux_for_Tegra/bootloader/tegraflash_internal.py", line 164, in print_process
>     outputchar = output.decode('ascii')
> UnicodeDecodeError: 'ascii' codec can't decode byte 0xe2 in position 0: ordinal not in range(128)
> Failed flashing t186ref.

Is 'ascii' the problem? Should it be 'utf-8' instead? Any advice on how to proceed would be appreciated.

pipex, 2022-10-20:
Hi there, thanks for reporting this. I found this forum topic on Nvidia that seems to be related
https://forums.developer.nvidia.com/t/flashing-xavier-nx-emmc/187379

Could you make sure that you are using `python2` as your default python interpreter, i.e. that the binary for `python` points to `python2` and try again?

Please let us know if that works for you

pipex, 2022-10-20:
Thank you for letting us know. I have created a PR updating the README instructions here #91

commit f62c3ae8 (2022-10-20): Document that Tegra flash requires python2

Trying to use the flasher with python3 may run into encoding issues as described by https://forums.developer.nvidia.com/t/flashing-xavier-nx-emmc/187379

Change-type: patch
Closes: #90
PR #91 (2022-10-20): Document that Tegra flash requires python2
Trying to use the flasher with python3 may run into encoding issues as described by https://forums.developer.nvidia.com/t/flashing-xavier-nx-emmc/187379

Change-type: patch
Closes: #90
closed 2022-10-20 by bulldozer-balena[bot]