balena-os/jetson-flash #99: Error while flashing Jetson AGX devkit

higmo, 2023-01-17:
I'm using the following Dockerfile:

```Dockerfile
FROM ubuntu:focal
ENV DEBIAN_FRONTEND=noninteractive
ENV TZ=Etc/UTC
RUN apt update
RUN apt install -y git nodejs npm usbutils zip
RUN git clone https://github.com/balena-os/jetson-flash.git
RUN cd jetson-flash/
RUN git config --global url."https://github".insteadOf git://github
RUN npm install
```

Which I'm building and running with:

```bash
docker build -t jetson-flash .
docker run -it --privileged -v /dev/bus/usb:/dev/bus/usb -v /path/to/img/folder:/work jetson-flash /bin/bash
```

I'm getting the following output when trying to flash from inside the container:

```bash
$ lsusb | grep NVIDIA
Bus 003 Device 002: ID 0955:7019 NVIDIA Corp. APX
$ ./bin/cmd.js -f /work/balena-cloud-agx-devkit-jetson-xavier-2.99.27+rev1-v14.0.8.img -m jetson-xavier
Consistency check done for /tmp/941/Linux_for_Tegra
Saved in /tmp/941
Successfully generated hash
Checking resin cache
Cache image check done
Extracting partition bootloader-dtb from /tmp/941/resin/img to /tmp/941/resin
Extracting partition bootloader-dtb_b from /tmp/941/resin/img to /tmp/941/resin
Extracting partition kernel from /tmp/941/resin/img to /tmp/941/resin
Extracting partition kernel_b from /tmp/941/resin/img to /tmp/941/resin
Extracting partition kernel-dtb from /tmp/941/resin/img to /tmp/941/resin
Extracting partition kernel-dtb_b from /tmp/941/resin/img to /tmp/941/resin
Extracting partition BMP from /tmp/941/resin/img to /tmp/941/resin
Extracting partition BMP_b from /tmp/941/resin/img to /tmp/941/resin
Extracting partition bpmp-fw-dtb from /tmp/941/resin/img to /tmp/941/resin
Extracting partition bpmp-fw-dtb_b from /tmp/941/resin/img to /tmp/941/resin
Extracting partition resin-boot from /tmp/941/resin/img to /tmp/941/resin
Extracting partition resin-rootA from /tmp/941/resin/img to /tmp/941/resin
Extracting partition resin-rootB from /tmp/941/resin/img to /tmp/941/resin
Extracting partition resin-state from /tmp/941/resin/img to /tmp/941/resin
Extracting partition resin-data from /tmp/941/resin/img to /tmp/941/resin
Data partition size in bytes: 212860928
Successfully generated hash
ODMDATA not specified, using default value set by BSP archive
(node:941) UnhandledPromiseRejectionWarning: undefined
(node:941) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:941) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

I've tried the few suggestions I found in other issues, like removing `/tmp/Linux_for_Tegra`, without success.

Any ideas? 🙂

alanb128, 2023-01-26:
@higmo I added a working Dockerfile to the Docker folder in this branch: https://github.com/balena-os/jetson-flash/tree/alanb128-doc - it will hopefully be merged shortly. Docker instructions are included in the README as well.

higmo, 2023-01-26:
Thanks a lot @alanb128! I ended up managing to flash my board with an Ubuntu Live after a couple hours of fiddling, but that'll be handy for next time!

closed 2024-09-24 by acostach