balena-os/jetson-flash #111: How to restore the Jetson Nano SD-CARD Devkit to the previous state, and the previous SD card cannot be turned on.

sanpis, 2023-04-23:
How to restore the Jetson Nano SD-CARD Devkit to the previous state, and the previous SD card cannot be turned on.

floion, 2023-04-24:
Hi. Can you detail on what the problem is and what you want to achieve?
The description you provided does not describe what is the problem you are facing with https://github.com/balena-os/jetson-flash

sanpis, 2023-04-27:
> Hi. Can you detail on what the problem is and what you want to achieve? The description you provided does not describe what is the problem you are facing with https://github.com/balena-os/jetson-flash

I installed opendatacam on Jetson with a new SD-CARD using this tool and it worked fine. But I switched back to the SD-CARD of the previous nvidia system and found that it could not be started. How can I do to restore the previous system？

floion, 2023-04-28:
Does your jetson sd nano devkit have an eMMC?

sanpis, 2023-05-01:
This is an official devkit, and without an eMMC.

sanpis, 2023-05-01:
> Does your jetson sd nano devkit have an eMMC?

This is an official devkit, and without an eMMC.

sanpis, 2023-05-03:
> Hi @sanpis,
> 
> > But I switched back to the SD-CARD of the previous nvidia system and found that it could not be started.
> 
> You mean that Nvidia Ubuntu cannot be started anymore if you plug in the SD-CARD with Nvidia Ubuntu, after you flashed balenaOS with jetson-flash?

Yes.

acostach, 2023-05-03:
@sanpis i's not possible to load Nvidia Ubuntu with the BalenaOS u-boot, which has been written by jetson-flash to the QSPI Flash when the device was provisioned. If you want to re-use the old Ubuntu sd-card you need to re-flash the QSPI using NVidia SDK Manager. You can use a third SD-CARD for that to ensure the original Ubuntu sd-card is preserved intact.

sanpis, 2023-05-04:
> @sanpis i's not possible to load Nvidia Ubuntu with the BalenaOS u-boot, which has been written by jetson-flash to the QSPI Flash when the device was provisioned. If you want to re-use the old [...]

It is very likely that this is the reason. Can the recovery function be added later, so that it will be more convenient for everyone to use.

closed 2024-09-24 by acostach