balena-os/jetson-flash #129: Jetson orin nano does not start

kyasby, 2023-10-15:
Hi,

I have a jetson-orin-nano-devkit-nvme, so I followd the README instruction.
Things goes well except the last step of the device apearing in the balena cloud.
No error appears while setting up.

How can I do with this?

acostach, 2024-01-31:
Hi, if you use a development variant of the image from balena-cloud, are you able to log in as root on the UART console and check if the board has ethernet or wifi connectivity?

csvke, 2024-02-22:
I have the same issue that Balena Cloud dashboard shows 'Post Provisioning' on the device.

I use the [Waveshare Jetson Orin Nano IO Board](https://www.waveshare.com/jetson-orin-io-base.htm) with a Jetson Orin Nano 4GB.

UART messages below:
`[0000.061] I> MB1 (version: 1.2.0.0-t234-54845784-562369e5)
[0000.066] I> t234-A01-1-Silicon (0x12347) Prod
[0000.070] I> Boot-mode : Coldboot
[0000.074] I> Entry timestamp: 0x00000000
[0000.077] I> last_boot_error: 0x0
[0000.080] I> BR-BCT: preprod_dev_sign: 0
[0000.084] I> rst_source: 0x0, rst_level: 0x0
[0000.088] I> Task: SE error check
[0000.092] I> Task: Bootchain select WAR set
[0000.096] I> Task: Enable SLCG
[0000.098] I> Task: CRC check
[0000.101] I> Task: Initialize MB2 params
[0000.106] I> MB2-params @ 0x40060000
[0000.109] I> Task: Crypto init
[0000.112] I> Task: Perform MB1 KAT tests
[0000.116] I> Task: NVRNG health check
[0000.120] I> NVRNG: Health check success
[0000.123] I> Task: MSS Bandwidth limiter settings for iGPU clients
[0000.129] I> Task: Enabling and initialization of Bandwidth limiter
[0000.135] I> No request to configure MBWT settings for any PC!
[0000.141] I> Task: Secure debug controls
[0000.145] I> Task: strap war set
[0000.148] I> Task: Initialize SOC Therm
[0000.152] I> Task: Program NV master stream id
[0000.156] I> Task: Verify boot mode
[0000.162] I> Task: Alias fuses
[0000.165] W> FUSE_ALIAS: Fuse alias on production fused part is not supported.
[0000.172] I> Task: Print SKU type
[0000.175] I> FUSE_OPT_CCPLEX_CLUSTER_DISABLE = 0x000001c8
[0000.180] I> FUSE_OPT_GPC_DISABLE = 0x00000002
[0000.184] I> FUSE_OPT_TPC_DISABLE = 0x000000fc
[0000.189] I> FUSE_OPT_DLA_DISABLE = 0x00000003
[0000.193] I> FUSE_OPT_PVA_DISABLE = 0x00000001
[0000.197] I> FUSE_OPT_NVENC_DISABLE = 0x00000001
[0000.202] I> FUSE_OPT_NVDEC_DISABLE = 0x00000000
[0000.206] I> FUSE_OPT_FSI_DISABLE = 0x00000001
[0000.210] I> FUSE_OPT_EMC_DISABLE = 0x0000000c
[0000.215] I> FUSE_BOOTROM_PATCH_VERSION = 0x7
[0000.219] I> FUSE_PSCROM_PATCH_VERSION = 0x7
[0000.223] I> FUSE_OPT_ADC_CAL_FUSE_REV = 0x2
[0000.227] I> FUSE_SKU_INFO_0 = 0xd6
[0000.230] I> FUSE_OPT_SAMPLE_TYPE_0 = 0x3 PS 
[0000.235] I> FUSE_PACKAGE_INFO_0 = 0x2
[0000.238] I> SKU: Prod
[0000.240] I> Task: Boost clocks
[0000.243] I> Initializing PLLC2 for AXI_CBB.
[0000.248] I> AXI_CBB : src = 35, divisor = 0
[0000.252] I> Task: Voltage monitor
[0000.255] I> VMON: Vmon re-calibration and fine tuning done
[0000.260] I> Task: UPHY init
[0000.265] I> HSIO UPHY init done
[0000.268] W> Skipping GBE UPHY config
[0000.272] I> Task: Boot device init
[0000.275] I> Boot_device: QSPI_FLASH instance: 0
[0000.280] I> Qspi clock source : pllc_out0
[0000.284] I> QSPI Flash: Macronix 64MB
[0000.288] I> QSPI-0l initialized successfully
[0000.292] I> Task: TSC init
[0000.295] I> Task: Load membct
[0000.298] I> RAM_CODE 0x4000441
[0000.301] I> Loading MEMBCT 
[0000.303] I> Slot: 0
[0000.305] I> Binary[1] block-3840 (partition size: 0x40000)
[0000.311] I> Binary name: MEM-BCT-1
[0000.314] I> Size of crypto header is 8192
[0000.318] I> Size of crypto header is 8192
[0000.322] I> strt_pg_num(3840) num_of_pgs(16) read_buf(0x40050000)
[0000.328] I> BCH of MEM-BCT-1 read from storage
[0000.333] I> BCH address is : 0x40050000
[0000.336] I> MEM-BCT-1 header integrity check is success
[0000.342] I> Binary magic in BCH component 1 is MEM1
[0000.346] I> component binary type is 1
[0000.350] I> strt_pg_num(3971) num_of_pgs(115) read_buf(0x40040000)
[0000.357] I> MEM-BCT-1 binary is read from storage
[0000.362] I> MEM-BCT-1 binary integrity check is success
[0000.367] I> Binary MEM-BCT-1 loaded successfully at 0x40040000 (0xe580)
[0000.374] I> RAM_CODE 0x4000441
[0000.379] I> RAM_CODE 0x4000441
[0000.383] I> Task: Load Page retirement list
[0000.387] I> Task: SDRAM params override
[0000.391] I> Task: Save mem-bct info
[0000.394] I> Task: Carveout allocate
[0000.397] I> RCM blob carveout will not be allocated
[0000.402] I> Update CCPLEX IST carveout from MB1-BCT
[0000.407] I> ECC region[0]: Start:0x0, End:0x0
[0000.411] I> ECC region[1]: Start:0x0, End:0x0
[0000.416] I> ECC region[2]: Start:0x0, End:0x0
[0000.420] I> ECC region[3]: Start:0x0, End:0x0
[0000.424] I> ECC region[4]: Start:0x0, End:0x0
[0000.428] I> Non-ECC region[0]: Start:0x80000000, End:0x80000000
[0000.434] I> Non-ECC region[1]: Start:0x0, End:0x0
[0000.439] I> Non-ECC region[2]: Start:0x0, End:0x0
[0000.444] I> Non-ECC region[3]: Start:0x0, End:0x0
[0000.448] I> Non-ECC region[4]: Start:0x0, End:0x0
[0000.459] E> BL_CARVEOUT: Failed to allocate memory of size 0x4000000 for CO:43.
[0000.467] C> Task 0x20 failed (err: 0x49490303)
[0000.471] E> Top caller module: BL_CARVEOUT, error module: BL_CARVEOUT, reason: 0x03, aux_info: 0x03
[0000.480] C> Boot Info Table status dump :
01111111001110001111111111111111`

csvke, 2024-02-22:
Is it because Balena does not support the 4GB version of Jetson Orin Nano?

acostach, 2024-02-22:
@csvke it seems that the boot process doesn't get further than MB1 and it could very likely be due to the different RAM module, as the module we test our images with is an 8GB RAM with SD-CARD, booting from the NVME. Do you have an 8GB model to check the image with?

csvke, 2024-02-22:
> @csvke it seems that the boot process doesn't get further than MB1 and it could very likely be due to the different RAM module, as the module we test our images with is an 8GB RAM with SD-CARD, booting from the NVME. Do you have an 8GB model to check the image with?

No I don’t unfortunately. Is there any thing I can change and build my own image?

I wish BalenaOS would clearly document that the 4GB version will NOT WORK out of the box

Manprocoder, 2024-04-24:
I used to install OS on the microSD Card with jetpack 5.1.2 and i encountered the error "Input not support" on display. Help me! Thanks a million.

Manprocoder, 2024-05-28:
> @Manprocoder <https://github.com/Manprocoder> sure, can you please let me
> know which type of Orin Nano module you are using and if the board shows up
> in balena-cloud after you provision it? By default Jetpack 5 does not
> support framebuffer display, but you can control the board trough
> balena-cloud, or trough the UART interface when using a development image,
> or, you can run X in a container like in this example

you're correct, i am using Jetpack 5.1.2. Could you send me a material link
or youtube link related to your means, please? Thanks a million.

acostach, 2024-05-28:
Hi, please see https://www.youtube.com/watch?v=bTGOzuZnOnk for Seeed Orin NX. For Orin Nano 8GB, please follow the steps in https://github.com/balena-os/jetson-flash?tab=readme-ov-file#orin-nx-flashing

If you have a Orin Nano 4GB, the Orin Nano 8GB is reported to not work on it. We are working to add Orin Nano 4GB support in balena-cloud.

acostach, 2024-06-12:
@csvke we now support the Seeed ReComputer J3010 which uses the 4GB Orin Nano module. That carrier board includes a custom change in the QSPI for HDMI, so I'm not sure if video output will work on the devkit but it's worth a try

acostach, 2024-09-23:
Closing due to inactivity, feel free to re-open if the issue still occurs with the J3010 image

closed 2024-09-23 by acostach