balena-os/jetson-flash #152: Installation of Balena0S on Nvidia Orin NX stuck at the end

gurbain, 2024-06-03:
Hello,

I have installed BalenaOS successsfully in the past on a RaspberryPI and on a Jetson Xavier AGX.

Nonetheless, I am quite stuck with the bootloading process for the Nvidia Orin NX. I have tried to follow the instructions at the letter. I am able to place the computer in recovery mode, to start and run the docker, the flashing process does not seem to show critical issue. Finally, **the LED switches off after around 15 minutes.** But when I switch my board off and on, **I am still on the previous OS version,** BalenaOS is not present. I tested on the Nvidia default dev kit and on another PC based on the orin NX module but I face the same issue on both

I really struggle to understand what is going on, since I cannot find a way to show the result of the flashing process from the USB to the Jetson QSPI memory.

Have anyone encountered the same problem and do you have any idea on how to get out of this? I thought that versions could be an issue but I am not sure.

- Host PC: Ubuntu 22.04.4 LTS (6.5.0-35-generic)
- Docker version 26.1.3, build b72abbb
- Hardware: Nvidia Jetson Orin NX 16GB
- BalenaOS: 5.3.10 (I cannot find on which L4T this is based)
- Jetson-flash: on master branch

Thanks you!

[flashing.log](https://github.com/user-attachments/files/15534858/flashing.log)

gurbain, 2024-06-12:
Hi @acostach,

Thank you for the reply. Since, I also opened the same discussion on the Balena discourse, I suggest to continue it there: https://forums.balena.io/t/installation-of-balena0s-on-nvidia-orin-nx-stuck-at-the-end/370686

As mentioned there, the boards I tested are:
- a [ZED Box - Orin NX 16GB for Edge AI | Stereolabs 1](https://store.stereolabs.com/en-eu/products/zed-box-nvidia-jetson-orin-nx-16gb)
- a [Rugged Computer RPC RML A4NX (GMSL) 1](https://www.syslogic.ai/jetson-orin-nx/rugged-computer-rpc-rml-a4nx-gmsl), basically based on a Orin NX 16GB module. The flashing procedure in their user manual completely refers to the flashing img, dtb and procedure of the dev kit.

acostach, 2024-06-12:
Alright, let's follow up in the forums then

closed 2024-06-12 by acostach