balena-os/jetson-flash #159: JetPack 6 and L4T v36.3 support for Jetson Orin AGX dev kit

volks73, 2024-06-17:
NVIDIA recently released the production version of JetPack 6 and L4T v36.3. This includes moving to Ubuntu 22.04 root file system and CUDA 12. 

Currently, Ubuntu 20.04 LTS is needed on the host device to flash JetPack 5. Ubuntu 20.04 is end-of-support or end-of-life at the end of 2024.

I would like to request support for flashing Jetson Orin AGX development kits with JetPack 6 and L4T v36.3.

volks73, 2024-07-03:
@acostach Thank you for the status update. This is greatly appreciated. @etohana created the original DTB for Jetpack v5. I have notified him that a new DTB will be needed for Jetpack v6. We will work on getting you a DTB file.

Just yesterday, July 2nd, 2024, I went through and updated balenaOS on our development fleet Orins to v5.3.21+rev3. Many of them were on v5.2.0 or v5.3.21 (no rev). Should I have upgraded to v5.3.21+rev2 first before v5.3.21+rev3? Basically, I skipped straight to v5.3.21+rev3 for all of the Orins in our development fleet.

 I cannot downgrade. Do I have to reflash all of the Orins with v5.3.21+rev2?

acostach, 2024-07-03:
@volks73 if they are development fleets and not production units, this means that if a device would fail to boot after the update, at most it would need to be re-flashed. But if the update works fine, there would be no need for re-flashing.

In my tests on the AGX Orin 32GB, Orin Nano 8GB and Orin NX SD, updating from 35.4.1 to 35.5.0 (on which your development devices are now) and then directly to 36.3 worked. The problem is that if the device didn't have both boot chains updated to 35.5.0 and if the update capsule failed to apply for any reason during this transition (i.e power cut, firmware didn't want to apply the capsule, etc), the device would not be able to rollback and would need to be re-flashed. The failure was somewhere in the MB2 firmware in this case. But if the capsule update went trough ok, or if both chains were running the latest L4T 35.5.0, the device did rollback successfully to v5.3.21+rev3 in case of any problem with the new OS.

So my suggestion would be to leave them as they are for now, and once the PR is merged and we have a draft release, I will provide you with the instructions for triggering the test update.

acostach, 2024-07-10:
Hi @volks73 the draft release v5.3.22 is now available for all the Orin device-types we support. You can test HUP on your lab devices using balena cli:

```
balena device os-update <device_uuid> --include-draft
? Target OS version 5.3.22-1720444904211 <-- this is the first draft release for Jetpack 6
? Host OS updates require a device restart when they complete. Are you sure you want to proceed? Yes
Updating the OS of wandering-tree to v5.3.22-1720444904211 [============            ] 50%
The device wandering-tree has been updated to v5.3.22-1720444904211 and will restart shortly!
```

Also, if you wish to download and flash the draft release you can use the following line:

```
curl 'https://api.balena-cloud.com/download?appId=<APP_ID>&deviceType=jetson-agx-orin-devkit&version=5.3.22&fileType=.zip'     -G     -H "Authorization: Bearer <TOKEN>"     --output ~/Downloads/balena-cloud-myapp-draft-jetson-agx-orin-devkit-v5.3.22.zip
```

Please note that the above curl may change when draft OS releases will be made available in the dashboard.

You can use the latest commit from the master branch of jetson-flash for provisioning the draft release, and also please make sure to check the notes in the device-repository [README](https://github.com/balena-os/balena-jetson-orin/?tab=readme-ov-file#linux-for-tegra-l4t-support) with regards to updating from L4T 35.5.0 to L4T 36.3.

Let me know how the update works for you and if you encounter any issue

volks73, 2024-07-15:
@acostach Great! Thank you working on this and the information. I test it out and get back to you, but it will be a couple of weeks before I get a chance to fully test due to travel obligations.

closed 2024-09-23 by acostach