balena-os/jetson-flash #176: Flashing Jetson AGX Orin Devkit 64GB

reinzor, 2024-08-02:
Hi there,

We are trying to flash a  Jetson AGX Orin Devkit 64GB device without any luck. I followed the steps of the README.md for the 32GB variant and my flashing process ends with:

```
*** The target generic has been flashed successfully. ***
Reset the board to boot from internal eMMC.
```

However, the device is not booting properly. I am getting a black screen. Is this supposed to work? Some other references I found:

- Open PR for the 64GB variant: https://github.com/balena-os/jetson-flash/pull/165 : seems incomplete
- A machine name `jetson-agx-orin-devkit-64gb` : https://docs.balena.io/reference/base-images/devicetypes/ however, this is not listed in with the following command:

```
$ balena devices supported | grep orin
jetson-agx-orin-devkit                                        aarch64 Nvidia Jetson AGX Orin Devkit 32GB
jetson-orin-nano-devkit-nvme                                  aarch64 Nvidia Jetson Orin Nano 8GB (SD) Devkit NVME
jetson-orin-nano-seeed-j3010                                  aarch64 Seeed reComputer J3010 4GB
jetson-orin-nx-seeed-j4012                                    aarch64 Seeed reComputer J4012 16GB
jetson-orin-nx-xavier-nx-devkit                               aarch64 Nvidia Jetson Orin NX 16GB in Xavier NX Devkit NVME
```

I am now simply using the `jetson-agx-orin-devkit` device type.

Any help is appreciated. Thanks!

myarmolinsky, 2024-08-02:
Unfortunately the Jetson AGX Orin Devkit 64GB is not currently supported. However, we are working on adding support for it at the moment and expect to have it done in about 4 weeks (this is just an estimate, we cannot guarantee that this is the timeline).
 
Sadly, I don't believe there's any way to install balena on this device properly at the moment, as the various jetson boards are quite differently set up than each other. For example, even the `AGX Orin 32GB` installation method won't work for the `AGX Orin 64GB`.

We should respond here again once support for the device type is properly added.

reinzor, 2024-08-02:
Thanks for the update!

acostach, 2024-08-09:
Hi @reinzor , the AGX Orin 64GB Devkit is now supported by balena and you can download OS images for it from balena-cloud. Please make sure to pull the latest changes in this repository and refer to https://github.com/balena-os/jetson-flash?tab=readme-ov-file#agx-orin-devkit-64gb-flashing for instructions on provisioning your device

closed 2024-08-09 by acostach

reinzor, 2024-08-13:
Great! Thanks for the quick follow-up!