balena-os/jetson-flash #194: Error flashing TX2 image with preload, image size limit?

mgabeler-lee-6rs, 2025-02-15:
Trying to use latest (v0.5.85) to flash a TX2 image with a preloaded release.

Preload was built by downloading the latest TX2 balenaOS image and then running:
```
balena preload --fleet $MYFLEET --commit $MYRELEASE --pin-device-to-release --splash-image ~/path/to/bootsplash....png balena-cloud-$FLEETNAME-jetson-tx2-6.3.19+rev1-v16.12.0.img
```
The resulting file is 12,788,179,968 bytes long (~12 GiB)

When I try to flash it, I get:
```
... License stuff ...
Retrieve BalenaOS image from /data/images/balena-cloud-$FLEETNAME-jetson-tx2-6.3.19+rev1-v16.12.0.img
/usr/src/app/jetson-flash/node_modules/balena-image-fs/node_modules/ext2fs/lib/libext2fs.js:1
... entire minified libext2fs source ...

ErrnoException: node_ext2fs_stat_i_size UNKNOWN (1107549184) args: [5662384]
    at ccallThrowAsync (/usr/src/app/jetson-flash/node_modules/balena-image-fs/node_modules/ext2fs/lib/util.js:22:9)
    at async Promise.all (index 8)
    at async /usr/src/app/jetson-flash/node_modules/balena-image-fs/node_modules/ext2fs/lib/fs.js:512:24
    at async /usr/src/app/jetson-flash/node_modules/balena-image-fs/node_modules/ext2fs/lib/util.js:84:11
    at async stat (/usr/src/app/jetson-flash/node_modules/balena-image-fs/node_modules/ext2fs/lib/fs.js:539:17) {
  errno: 1107549184,
  syscall: 'node_ext2fs_stat_i_size',
  code: 'UNKNOWN'
}

Node.js v18.20.6
```

However, if I try to flash the original balenaOS image I downloaded _without_ the preload (2,378,170,368 bytes), it works fine.

acostach, 2025-02-17:
Hi, the failure actually comes from https://github.com/balena-io-modules/node-ext2fs and it seems this error is thrown when the non-flasher image's size is being determined.

mgabeler-lee-6rs, 2025-02-20:
@acostach The linked fix to node-ext2fs has gotten me past the size isssue!

The flash image prep still fails, but with an unrelated issue for which I've filed #195, and I have an easy enough workaround for that.

closed 2025-02-21 by acostach